openXsensor/openXsensor #12: BMP180 pressure sensor support ?

Sovxx, 2015-12-03:
Hi,
I found out this sensor, which is significantly cheaper than the MS5611.
It would be nice if it could be supported too ;)

nigelsheffield, 2015-12-05:
Bmp180 already supported!
It is significantly cheaper then ms5611,
Is also significantly less good lol.
It's OK for altitude, vario tones are not so good, usable and better then frsky normal precision vario though.
Ms5611 can be had for £5 now so not bank breaking but bmp180 is so cheap now £0.60  it could be added to a sensor that you were not too bothered about the accuracy.

I might try it along with the IMU gy-521 at some point , it might work pretty well maybe?

mstrens, 2015-12-08:
BMP180 is already supported

closed 2015-12-08 by mstrens