openXsensor/openXsensor #16: GPS not showing up on my Taranis

Sovxx, 2016-03-20:
Hi,

I don't know much to Arduino programming but in the past, I tried the voltmeter, the altimeter and the ampmeter of openXsensor which all worked great !

Unfortunately, my GPS setup doesn't work :

I have an Arduino Uno connected to a GY-NEO6MV2 module
with the TX pin of the GPS connected to pin 0
the RX pin of the GPS connected to pin 6
and my X8R smart port connecte to pin 4.

The "discover new sensors" function of the Taranis won't show up anything...

I uncommented the following lines in oXs_gps.cpp :
```
#ifdef DEBUG
#define DEBUGGPS
#define DEBUGPARSEGPS
#define DEBUGSENDGPS
#define DEBUG_FORWARD_GPS_MSG_TO_PC
#endif
```

The serial monitor in the Arduino IDE (38400bps) keeps showing this very sequence every second :
```
...
0
0
0
,0,80,80,F8,0,78,78,F8,0,78,0,78,0,78,78,0,80,0,80,78,80,0,F8,80,0,F8,80,80,80,80,80,78,0,80,0,80,F8,80,80,0,78,0,78,0,78,0,78,0,78,0,80,0,78,0,80,0,78,0,78,0,78,78,80,78,78,0,78,78,80,F8,0,78,0,78,80,78,80,0,F8,80,0,80,78,0,80,78,0,80,80,F8,0,78,78,F8,0,78,80,78,78,78,78,78,78,78,78,80,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,0,78,80,78,80,78,80,78,0,78,F8,0,80,78,0,80,80,F8,0,78,78,F8,0,78,80,78,80,78,78,78,80,80,80,0,78,78,80,0,78,0,80,78,F8,80,0,78,0,80,80,F8,80,0,78,0,78,0,78,78,80,0,80,0,78,80,80,80,0,F8,80,78,0,0,78,78,80,0,80,78,78,80,80,78,0,78,78,78,80,F8,0,78,80,F8,0,0,78,78,80,80,80,80,80,F8,F8,0,F8,80,0,80,F8,78,78,F8,78,80,80,78,78,80,80,0,80,80,78,0,80,78,80,78,80,78,80,78,0,0,78,80,80,80,80,78,F8,F8,80,0,80,0,80,F8,78,78,80,78,80,0,78,0,80,0,80,80,0,80,F8,80,80,0,80,0,80,80,80,78,78,78,0,78,78,0,0,78,80,80,0,80,0,80,78,0,80,78,0,80,80,78,0,80,F8,80,80,80,0,80,80,78,0,0,78,80,78,78,0,80,80,0,800
0
0
0
...

```
Do you have any idea ?
Thanks a lot

mstrens, 2016-03-23:
That is strange. Normally latest version of oXs works with GPS and SPORT.
I made a version last week and I got GPS data on my Taranis.
Still, please note that GPS data are sent only when GPS got a fix (see enough satelites). This can take some minutes and best is to try outdoor.
On my NEO6, I have a led that blink when GPS has a fix.

Still, this is probably not the reason for you issue.
If you get data on the PC every second instead of every 200 msec, I expect that there is an issue when oXs tries to initialise you GPS.
 In fact at power on, oXs expect that the GPS is running his default config (e.g. 9600 bds, data delivered at 1Hz, NMEA format). oXs sent then a set of command to force the GPS to switch to other parameters (38400 bds, 5hz, UBLOX format, ...).
It could be that your GPS default set up is not the expected one and so the commands sent by oXs fails.
I think many GPS module with a neo6 have an eeprom that let save some changes to the default config of the GPS. I expect that your device has been loaded with a "non standard" default set up.
You can then change back this default set up using the program delivered by UBLOX.

Here an extract of the instructions I put in the documentation.
* Software points of attention
*    UBLOX GPS module are normally delivered with a default configuration (generating automatically e.g some NMEA messages at 9600 bauds at a refresh rate of 1 hz).
*    oXs assumes that, at start up, GPS is working at 9600 bauds. oXs sent then some commands in order to
*       - disable all NMEA messages
*       - activates some UBX messages
*       - increase frequency of calculation (to 5 Hz instead of 1hz)
*       - set up the baud rate to 38400 instead of 9600.
*    Those parameters are not saved in the GPS (because some GPS modules does not allow it). So, oXs will send those commands at each power on.   
*    If you oXs does not send GPS data, please check that your GPS module has still the default configuration (most important is that it is configured to receive UBX command messages at 9600 bauds). 
*       An easy way to check the GPS configuration is to connect the GPS module to a 3.3 volt FTDI ( or USB to serial adapter) and to use a free software named "u-center". 
*       This software is available on the official web site of UBLOX. More info is easily available on the web. 

If you use a FTDI to change the setup, take care to use a 3.3Volt version.

I suggest that, if you need some more help, you ask your questions on http://openrcforums.com/ in the openxvario section. I do not look quite often at the issues reported on github but I read normally daily openrcforums.com

Sovxx, 2016-04-02:
Hi,
thanks for the reply.
I have not tried troubleshooting more since this time.
I don't have a FTDI to test.

I am using https://github.com/SamuelBrucksch/diy-frsky-gps which works fine
Thanks again

closed 2017-02-19 by mstrens