openXsensor/openXsensor #20: oXs ver.7.0 MPX protocol GPS

nostromo123, 2016-09-25:
Just did a desktop test with GPS under MPX protokol.

[image]

This is the output part in  oXs_config.h 

_// ***** 9.3 - Multiplex data *****
#define SETUP_MULTIPLEX_DATA_TO_SEND    \
                        4 , GPS_COURSE , 1 , 1 , 0 , -16384 , 16383 , \
                        5 , GPS_SPEED , 1 , 1 , 0 , -16384 , 16383 , \
                        6 , GPS_ALTITUDE , 1 , 1 , 0 , -16384 , 16383 , \
                        7 , GPS_DISTANCE , 1 , 1 , 0 , -16384 , 16383 , \
                        8 , GPS_BEARING , 1 , 1 , 0 , -16384 , 16383_

[image]
[image]

**In postition #7 on display should it not be 0.7 _m_ instead of 0.7 _km_?**

br KH

nostromo123, 2016-09-26:
Could be, that km is ok, but then this line in oXs_config_description.h might be wrong

*    GPS_DISTANCE      0.1 m     Distance from home 

Must be km instead of m here??

mstrens, 2017-02-17:
oXs calculates gps distance in m.
There is a bug in a line from oXs_multiplex.h
In V7 and before, there is 
                        MU_VOLT , MU_CURR , MU_MAH , MU_DIR , MU_ASPD , MU_ALT , MU_RPM , MU_DIST , MU_DIR ,  MU_LEVEL ,\
It should be 
                        MU_VOLT , MU_CURR , MU_MAH , MU_DIR , MU_ASPD , MU_ALT , MU_RPM , MU_ALT , MU_DIR ,  MU_LEVEL ,\ 

I will fix it in next version (= version 8)

closed 2017-02-17 by mstrens