openXsensor/openXsensor #23: openXsensor2.5  not compatible with latest configurator 3.0

Ryanf55, 2017-01-11:
[image]

I got this error when trying to write my config. I have downloaded the latest OX sensor and configurator from the downloads section. 

Thanks!

davxlw, 2017-01-15:
Unfortunately, there is not an openXsensor v3.0 official release in the download section :(
But you can get it through this link: https://github.com/openXsensor/openXsensor/archive/OpenXsensor_Multiplex_v3.0.zip

Ryanf55, 2017-02-03:
Thanks! 
I downloaded that file and used the configurator in the downloads section. 
[image]

I can't get the variometer working. I have used/sold KK variometers, and wanted to try the OpenTX version. It doesn't give me any data when using the configurator. 

However, if I flash the arduino default sketch, it does work as a variometer, but obviously is not configurable. 

Here's my config file, using a D4R II receiver, sending data from arduino D4 to RX pin of the D4R ii receiver. 

Thanks!

mstrens, 2017-02-17:
Sorry but the configurator is not maintained anymore.
So currently you have to edit manually the file oXs_config.h.
All explanations are provided in the file oXs_config_description.h

closed 2017-02-17 by mstrens