openXsensor/openXsensor #26: Problems with connection to the receiver / receiver could not find telemetry

labeh, 2017-04-11:
My TX (Horus and also Taranis) could not find OXS Sensor in Sensor Discovery when I plug them into my X6R (tried also another X6R). I pulled the latest Master branch and also tried various other tagged versions from GitHub.

My Hardware is an Arduino Nano which should be ok, because everything (blink, output test, etc.) works without problems. When I have the software on it and enable debug I also could log a proper startup:
`openXsensor starting..
 milli=0
freeRam=1590
FRSky Output Module: TX Pin=4
Sport protocol= 1
End of general set up
`
My Flashing Hardware is Ubuntu 16.04 and Arduino IDE 1.8.1.

What should I do to get OXS working?

mstrens, 2017-04-18:
I suggest that you send a copy of your oXs_config.h file and that you explain which kind of sensors are connected (MS5611, ....).
I suggest that we continue this support on this forum: 
http://openrcforums.com/forum/viewforum.php?f=86
because I do not look always at github issues.

On this forum, you can create a new topic. There are also other members that can help you

labeh, 2017-07-24:
`// OpenXsensor https://github.com/openXsensor/
// started by Rainer Schlosshan and maintained by Michel Strens

// This is version : 8.0.6 (8 march 2017)

//******************************************************************************************************************************************************* //
//                                                                                                                                                        //
// ! IMPORTANT NOTE !!!!!!!!!!   Another file in this project provides detailed explanations on how to set up this file (see oXs_config_description.h).   //
//                                                                                                                                                        //
//******************************************************************************************************************************************************* //
//  Note: the oXs_config.h file present on the GitHub site is not always meaningful.
//       It could be that the combination of active/non active parameters is not consistent.
//       This is just the result of many updates and tests in this document.
//       So take always care to set up the oXs_config.h file according to your needs and check carefully all options.
//  For old version 2 and 3 of oXs, there was a program named "configurator" running on PC that allowed easier editing of this oXs_config.h file
//       Still this configurator is not developped anymore. So currently, this oXs_config.h file has to be edited manually. 

#ifndef OXS_CONFIG_h
#define OXS_CONFIG_h

// --------- 1 - Telemetry protocol ---------
#define PROTOCOL  FRSKY_SPORT  // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

// --------- 2 - Serial data pin choice ---------
#define PIN_SERIALTX      4    // The pin which transmits the serial data to the telemetry receiver, Usually pin 4 (otherwise pin 2)

// --------- 3 - PPM settings ---------
#define PIN_PPM           3       // default is 2 but my own device use 3
#define PPM_MIN_100       980     // default 1500 - 512 ; // pulse width (usec) when TX sends a channel = -100
#define PPM_PLUS_100      1990    // default 1500 + 512 ; // pulse width (usec) when TX sends a channel = +100

// --------- 4 - Vario settings ---------

// ***** 4.1 - Connecting 1 or 2 barometric sensor(s)  *****
//#define VARIO // set as comment if there is no vario
//#define SENSOR_IS_BMP180 // set as comment if first baro sensor is a BMP180 or BMP085 (instead of a MS5611)
//#define SENSOR_IS_BMP280 // set as comment if first baro sensor is a BMP280 (instead of a MS5611)
//#define VARIO2 // set as comment if there is no second vario

// ***** 4.2 - Sensitivity predefined by program *****
#define SENSITIVITY_MIN 50
#define SENSITIVITY_MAX 300
#define SENSITIVITY_MIN_AT 100
#define SENSITIVITY_MAX_AT 1000

// ***** 4.3 - Sensitivity adjusted from the TX *****
#define SENSITIVITY_MIN_AT_PPM 10    // sensitivity will be changed by OXS only when PPM signal is between the specified range enlarged by -5 / +5
#define SENSITIVITY_MAX_AT_PPM 40
#define SENSITIVITY_PPM_MIN  20      // common value for vario is 20
#define SENSITIVITY_PPM_MAX 100      // common value for vario is 100

// ***** 4.4 - Hysteresis parameter & Alt compensation based on sensor temp *****
#define VARIOHYSTERESIS 5
//#define ALT_TEMP_COMPENSATION 800

// ***** 4.5 - Vertical speeds calculations *****
#define VARIO_PRIMARY       FIRST_BARO          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define VARIO_SECONDARY     AIRSPEED_COMPENSATED          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define SWITCH_VARIO_MIN_AT_PPM 10
#define SWITCH_VARIO_MAX_AT_PPM 90

// ***** 4.6 - Analog vertical speed *****
//#define PIN_ANALOG_VSPEED 3
#define ANALOG_VSPEED_MIN -3
#define ANALOG_VSPEED_MAX  3

// ***** 4.7  - Calculating glider ratio, average sink/climb rate ******************************
//#define GLIDER_RATIO_CALCULATED_AFTER_X_SEC 1  // value must be higher or equal to 1, set line as comment if no calculation must be performed
#define SPEED_TOLERANCE   5              // in % of speed
#define VSPEED_MIN_TOLERANCE -200        // out of tolerance when Vspeed is lower than this value  (cm/sec)
[... 140 lines cut ...]
  int32_t value ;
} ;

#define FRSKY_SPORT      1
#define FRSKY_HUB        2
#define FRSKY_SPORT_HUB  3
#define MULTIPLEX        4
#define HOTT             5
#define JETI             6

#define SECONDS_SINCE_T0        32
#define AVERAGE_VSPEED_SINCE_TO 33 

#ifdef DEBUG
//#include "HardwareSerial.h"
#endif

#ifdef GPS_INSTALLED
//#include "HardwareSerial.h"
#endif

#endif// End define OXS_CONFIG_h
`

I have only a pure Arduino Nano attached to the receiver. No sensor is present except of the "voltage sensor". But it is not connected to the battery.

mstrens, 2017-07-24:
In order to get a voltage from oXs, you have :
- to uncomment this line and says which analog arduino pin you are using.
//#define PIN_VOLTAGE 2 , 3 , 1 , 8 , 8 , 8 // set this line as comment if no one voltage have to be measured, set the value to 8 for the voltage(s) not to be measured.

So line could be the following if you want to measure volatge on pin A2
#define PIN_VOLTAGE 2 , 8 , 8 , 8 , 8 , 8 // set this line as comment if no one voltage have to be measured, set the value to 8 for the voltage(s) not to be measured.

- to specify in which telemetry field the voltage has to be sent. So you should have a line like
#define VFAS_SOURCE VOLT_1 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
Note: I uncommented this line in your config (I remove the // in front of the line). So oXs should transmit VOLT1 (= voltage on first pin defined so here on A2) as VFAS.

If this is unclear let me know.

mstrens, 2017-07-24:
Did you read the file oXs_config_description.h
This file explains how to edir the oXs_config.h file

ucakar, 2022-01-26:
Hi

My TX (FrSky TANDEM X20 AND Taranis Lite S) could not find OXS Sensor in Sensor Discovery when I plug them into my Archer GR8 Sport and 9MX Spport.. I pulled the latest OpenXsensor v8.0.1 GitHub.

My Hardware is an Arduino Nano which should be ok, because everything (blink, output test, etc.) works without problems.

mstrens, 2022-01-27:
There are plenty of oXs_sensor in use. So normally the firmware should be ok.
There is probably something wrong in your configuration files.
I propose that you put here your 2 config files (oXs_config_basic.h and oXs_config_advanced.h) and that you said what are the sensors connected on the arduino nano.

ucakar, 2022-01-27:
Hi,

As I am using version 8.0.1 I can not find oXs_config_advanced.h only basic. which I am attaching here.

// OpenXsensor https://github.com/openXsensor/
// started by Rainer Schlosshan and maintained by Michel Strens

// This is version : 8.02 (19 february 2017)

//******************************************************************************************************************************************************* //
//                                                                                                                                                        //
// ! IMPORTANT NOTE !!!!!!!!!!   Another file in this project provides detailed explanations on how to set up this file (see oXs_config_description.h).   //
//                                                                                                                                                        //
//******************************************************************************************************************************************************* //
//  Note: the oXs_config.h file present on the GitHub site is not always meaningful.
//       It could be that the combination of active/non active parameters is not consistent.
//       This is just the result of many updates and tests in this document.
//       So take always care to set up the oXs_config.h file according to your needs and check carefully all options.
//  For some previous version of oXs, there was a program named "configurator" running on PC that allowed easier editing of this oXs_config.h file
//       Still this configurator is not developped anymore. So currently, this oXs_config.h file has to be edited manually. 

#ifndef OXS_CONFIG_h
#define OXS_CONFIG_h

// --------- 1 - Telemetry protocol ---------
#define PROTOCOL  FRSKY_SPORT  // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

// --------- 2 - Serial data pin choice ---------
#define PIN_SERIALTX      4    // The pin which transmits the serial data to the telemetry receiver, Usually pin 4 (otherwise pin 2)

// --------- 3 - PPM settings ---------
//#define PIN_PPM           3       // default is 2 but my own device use 3
#define PPM_MIN_100       980     // default 1500 - 512 ; // pulse width (usec) when TX sends a channel = -100
#define PPM_PLUS_100      1990    // default 1500 + 512 ; // pulse width (usec) when TX sends a channel = +100

// --------- 4 - Vario settings ---------

// ***** 4.1 - Connecting 1 or 2 MS5611 barometric sensor *****
//#define VARIO // set as comment if there is no vario
//#define SENSOR_IS_BMP180 // set as comment if baro sensor is MS5611 and BMP180 or BMP085 is used
//#define SENSOR_IS_BMP280 // set as comment if baro sensor is MS5611 and BMP280 is used
//#define VARIO2 // set as comment if there is no second vario

// ***** 4.2 - Sensitivity predefined by program *****
#define SENSITIVITY_MIN 50
#define SENSITIVITY_MAX 300
#define SENSITIVITY_MIN_AT 100
#define SENSITIVITY_MAX_AT 1000

// ***** 4.3 - Sensitivity adjusted from the TX *****
#define SENSITIVITY_MIN_AT_PPM 10    // sensitivity will be changed by OXS only when PPM signal is between the specified range enlarged by -5 / +5
#define SENSITIVITY_MAX_AT_PPM 40
#define SENSITIVITY_PPM_MIN  20      // common value for vario is 20
#define SENSITIVITY_PPM_MAX 100      // common value for vario is 100

// ***** 4.4 - Hysteresis parameter & Alt compensation based on sensor temp *****
#define VARIOHYSTERESIS 5
//#define ALT_TEMP_COMPENSATION 800

// ***** 4.5 - Vertical speeds calculations *****
#define VARIO_PRIMARY       FIRST_BARO          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define VARIO_SECONDARY     FIRST_BARO          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define SWITCH_VARIO_MIN_AT_PPM 10
#define SWITCH_VARIO_MAX_AT_PPM 90

// ***** 4.6 - Analog vertical speed *****
//#define PIN_ANALOG_VSPEED 3
#define ANALOG_VSPEED_MIN -3
#define ANALOG_VSPEED_MAX  3

[... 141 lines cut ...]
#include <Arduino.h>
struct ONE_MEASUREMENT {
  uint8_t available ;
  int32_t value ;
} ;

#define FRSKY_SPORT      1
#define FRSKY_HUB        2
#define FRSKY_SPORT_HUB  3
#define MULTIPLEX        4
#define HOTT             5
#define JETI             6

#define SECONDS_SINCE_T0        32
#define AVERAGE_VSPEED_SINCE_TO 33 

#ifdef DEBUG
//#include "HardwareSerial.h"
#endif

#ifdef GPS_INSTALLED
//#include "HardwareSerial.h"
#endif

#endif// End define OXS_CONFIG_h

mstrens, 2022-01-27:
Which sensors did you connected to arduino?
It looks like you have a GPS and a ADS1115. Is this rigth?
It looks like you want also to measure 3 voltages directly on 3 pins from arduino (on top of ADS1115). Why?
Note: the resistor values for "Voltages" seems me a bit strange strange if you want to measure lipo cells. 

I strongly suggest that you use the latest version of the firmware.

ucakar, 2022-01-27:
Hi,

I downloaded the latest version from here: https://openxsensor.github.io

I am using GPS B220 only, no other sensors, I do not want to measure LiPo cells, I only want to have telemetry data of GPS sensor.

Can you send me link to latest firmware?

mstrens, 2022-01-27:
The latest version is in the master branch of https://github.com/openXsensor/openXsensor
For a GPS, you should make some changes to oXs_config.basic.h file (not to the oXs_config.advanced.h).
I think the changes are easy to understand.

Please note that oXs_sensor starts sending GPS data only when the arduino got a fix from the GPS. It can take several minutes (depending on GPS position) before the GPS data will be "discovered" on the Tx. There is usually a led on the GPS that show when the GPS got the FIX. Even when this led shows a fix, it can still take a few seconds more before getting the data on TX because there are several levels of fixes (and I thing that oXs starts only when the most accurate fix level is available).

ucakar, 2022-01-27:
Hi,

I downloaded latest master version from github and from basic configuration selected GPS only, here is my configuration basic.h but still not discoverd sensor eventhoug on my gps is LED blinking that 3D fix is stable.

// OpenXsensor https://github.com/openXsensor/
// started by Rainer Schlosshan and maintained by Michel Strens

// This is version : 8.2.15 (27 Oct 2020)

//******************************************************************************************************************************************************* //
//                                                                                                                                                        //
// ! IMPORTANT NOTE !!!!!!!!!!   Another file in this project provides detailed explanations on how to set up this file (see oXs_config_description.h).   //
//                                                                                                                                                        //
//******************************************************************************************************************************************************* //
//  Note: the oXs_config_basic.h and oXs_config_advanced.h files present on the GitHub site are not always meaningful.
//       It could be that the combination of active/non active parameters is not consistent.
//       This is just the result of many updates and tests in this document.
//       So take always care to set up the files according to your needs and check carefully all options.
//  For old version 2 and 3 of oXs, there was a program named "configurator" running on PC that allowed easier editing of a oXs_config.h file
//       Still this configurator is not developped anymore. So this version request manual editing. 

#ifndef OXS_CONFIG_BASIC_h
#define OXS_CONFIG_BASIC_h

// --------- 1 - Telemetry protocol ---------  
#define PROTOCOL  FRSKY_SPORT   // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

// ***** 1.1 - Pin connected to Rx ************                              See oXs_config_advanced.h (normally no need to change it) (default is arduino pin 4 connected to RX) 
// ***** 1.2 - SPORT_SENSOR_ID (used only for Frsky Sport protocol)  *****   See oXs_config_advanced.h (normally no need to change it) 

// --------- 2 - Data to transmit ---------   uncomment the lines below if you want to transmit some extra oXs measurements into some telemetry fields
// ***** 2.1 - Frsky data *****                
//#define VFAS_SOURCE  VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4 
//#define FUEL_SOURCE   ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define A3_SOURCE     ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define A4_SOURCE     ADS_VOLT_3                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ACCX_SOURCE     TEST_1                   //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4 
//#define ACCY_SOURCE     TEST_2                 //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ACCZ_SOURCE     TEST_3                 //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define T1_SOURCE       TEST_1                   //  select between TEST_1, TEST_2, TEST_3 , GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , SENSITIVITY, PPM, VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define T2_SOURCE       TEST_2                    //  select between TEST_1, TEST_2, TEST_3, , GLIDER_RATIO, SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , SENSITIVITY, PPM, VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4

// ***** 2.2 - Hott data *****
//#define CELL_UNDERVOLTAGE_WARNING 3300            // Warning threshold in mV; 
//#define BATTERY_1_SOURCE          ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define BATTERY_2_SOURCE          ADS_VOLT_2                 // select between VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define MAIN_BATTERY_SOURCE       ADS_VOLT_3                 // select between VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define TEMPERATURE_1_SOURCE      VOLT_1          //  select between VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, TEST_1, TEST_2, TEST_3 , GLIDER_RATIO , SENSITIVITY , PPM
[... cut ...]
#define A_LOCATOR_IS_CONNECTED      NO          // select between YES , NO

// --------- 13 - Rf link quality ---------------                                  see oXs_config_advanced.h for additionnal parameters about link quality                                  
#define MEASURE_RF_LINK_QUALITY     NO         // select between YES , NO

// --------- 20 - Sequencer ---------                                               see oXs_config_advanced.h (only when oXs has to generate signals in sequence)

#endif// End define OXS_CONFIG_BASIC_h

mstrens, 2022-01-27:
Your config seems ok.
In order to find the reason of the issue, I propose that you change the config in order to to let your Tx discover also another sensor.
Therefore, I suggest that you ask also for VFAS.
So in config_basic you must  
- uncomment the line //#define VFAS_SOURCE VOLT_1 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
#define VFAS_SOURCE VOLT_1 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
- have #define ARDUINO_MEASURES_VOLTAGES YES // select between YES , NO (When NO, following line is discarded)

When those changes are done, the TX should discover "Vfas" even if the value should be "0" (because there is no voltage on the arduino pin defined in config_advanced
For this test, there is no need to change you hardware.
If you get Vfas on Tx, it shows at least that oXs can communicate with the RX and the issue is GPS related.
If not it shows that the issue is not GPS related.

ucakar, 2022-01-27:
I did it. but no VFAS sensor found :(

Do I have to put some resistor between pin 4 and signal cable that goue to smart port on receiver?

ucakar, 2022-01-27:
I am using receiver Archer GR8 using ACCESS and board is Arduino Pro mini 328 16MHz 5V. could be that problem?

mstrens, 2022-01-27:
Arduino pro mini  (8 or 16 Mhz) is for sure not a problem.
I do not have a receiver with Access protocol but your receiver has a S port pin.
So I expect it should work with oXs but I never test it myself.
Perhaps can you ask on a forum if someone use oXs with an access protocol.

mstrens, 2022-01-27:
> I did it. but no VFAS sensor found :(
> 
> Do I have to put some resistor between pin 4 and signal cable that goue to smart port on receiver?

There is no need for a resistor.

ucakar, 2022-01-27:
> Arduino pro mini (8 or 16 Mhz) is for sure not a problem.
> I do not have a receiver with Access protocol but your receiver has a S port pin.
> So I expect it should work with oXs but I never test it myself.
> Perhaps can you ask on a forum if someone use oXs with an access protocol.

I will borrow 1 X6R or X8R receier with ACCST and report here. Thanks for advice!

mstrens, 2022-01-28:
I had a look on internet for Rx with access protocol.
It seems that openTx has some menu to setup options for those receivers.
Those options allows to activate/deactivate Telemetry and to select between S.port and F.port telemetry protocol.

To let oXs works, you have to activate telemetry and NOT activate F.Port
Here a link to a video abour archer Rx.
https://www.youtube.com/watch?v=ZSnqvBh4aK0

ucakar, 2022-01-28:
Hello,

Yes I know that option from my Taranis X Lite Pro, I went to my receiver options on transmitter and allready checked out that I am using S-Port.

I also mapped on my R9SX receiver to be on pin5.

So no idea anymore...

Yesterday I borrowed S8R receiver from my colleague, but same problem, I can not find sensors...eventhough is working on ACCST D16.

Today I will borrow X8R receiver that should be supported...and I hope I will find solution :)

Thanks for all help.

Am I only with that kind of problem?

mstrens, 2022-01-28:
oXs should work with any accst receiver (or an access with S.port enabled).
So there must be an issue with:
- your soldering/wiring
- the config files
- your arduino board.

Did you connect oxs with 3 wires to theRx (Gnd, 5V, S.port attached to arduino pin "4")
You can post here your 2 config files (used with version in master branch) so I can check them.

ucakar, 2022-01-28:
Hi,

Here are pics:

[img]http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/1.jpg[/img]

[img]http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/1.jpg[/img]

[img]http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/3.jpg[/img]

[img]http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/4.jpg[/img]

[img]http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/5.jpg[/img]

ucakar, 2022-01-28:
here is basic configuration file:

http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/oXs_config_basic.h

here is advanced configuration file:

http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/oXs_config_description.h

mstrens, 2022-01-28:
Using a chinese copy of arduino should not be an issue.
Still I got once a an issue:  I ordered 5V 16mhz but I got in fact a 3.3V 8mHz version.
I did not noticed it immediately because I was able to flash it selecting the 16Mhz board in arduino IDE but the program (it was not oXS) did not run as expected. I measured the voltage on Vcc pin (not Raw) and I noticed that it was 3.3V so probably a 8Mhz version. Once I flashed it selecting a 8Mz board in the arduino IDE it works OK.

Note that the config_desciption.h file is just for information. The 2 files that really matter are config_basic.h and config_advanced.h

I suggest that you make a test sending only VFAS. To do so, you can replace oXs_config_basic.h with the lines here below (I activate Vfas, Voltage and disabled GPS:

// OpenXsensor https://github.com/openXsensor/
// started by Rainer Schlosshan and maintained by Michel Strens

// This is version : 8.2.15 (27 Oct 2020)

//******************************************************************************************************************************************************* //
//                                                                                                                                                        //
// ! IMPORTANT NOTE !!!!!!!!!!   Another file in this project provides detailed explanations on how to set up this file (see oXs_config_description.h).   //
//                                                                                                                                                        //
//******************************************************************************************************************************************************* //
//  Note: the oXs_config_basic.h and oXs_config_advanced.h files present on the GitHub site are not always meaningful.
//       It could be that the combination of active/non active parameters is not consistent.
//       This is just the result of many updates and tests in this document.
//       So take always care to set up the files according to your needs and check carefully all options.
//  For old version 2 and 3 of oXs, there was a program named "configurator" running on PC that allowed easier editing of a oXs_config.h file
//       Still this configurator is not developped anymore. So this version request manual editing. 

#ifndef OXS_CONFIG_BASIC_h
#define OXS_CONFIG_BASIC_h

// --------- 1 - Telemetry protocol ---------  
#define PROTOCOL  FRSKY_SPORT   // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

// ***** 1.1 - Pin connected to Rx ************                              See oXs_config_advanced.h (normally no need to change it) (default is arduino pin 4 connected to RX) 
// ***** 1.2 - SPORT_SENSOR_ID (used only for Frsky Sport protocol)  *****   See oXs_config_advanced.h (normally no need to change it) 

// --------- 2 - Data to transmit ---------   uncomment the lines below if you want to transmit some extra oXs measurements into some telemetry fields
// ***** 2.1 - Frsky data *****                
#define VFAS_SOURCE  VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4 
//#define FUEL_SOURCE   ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define A3_SOURCE     ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define A4_SOURCE     ADS_VOLT_3                 // select between VOLT_1, VOLT_2, VOLT_3 , VOLT_4, VOLT_5 , VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ACCX_SOURCE     TEST_1                   //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4 
//#define ACCY_SOURCE     TEST_2                 //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ACCZ_SOURCE     TEST_3                 //  select between TEST_1, TEST_2, TEST_3, GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, PITCH, ROLL , YAW, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define T1_SOURCE       TEST_1                   //  select between TEST_1, TEST_2, TEST_3 , GLIDER_RATIO , SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , SENSITIVITY, PPM, VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define T2_SOURCE       TEST_2                    //  select between TEST_1, TEST_2, TEST_3, , GLIDER_RATIO, SECONDS_SINCE_T0 ,AVERAGE_VSPEED_SINCE_TO , SENSITIVITY, PPM, VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4

// ***** 2.2 - Hott data *****
//#define CELL_UNDERVOLTAGE_WARNING 3300            // Warning threshold in mV; 
//#define BATTERY_1_SOURCE          ADS_VOLT_1                 // select between VOLT_1, VOLT_2, VOLT_3, VOLT_4, VOLT_5, VOLT_6, ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
[... cut ...]
#define A_LOCATOR_IS_CONNECTED      NO          // select between YES , NO

// --------- 13 - Rf link quality ---------------                                  see oXs_config_advanced.h for additionnal parameters about link quality                                  
#define MEASURE_RF_LINK_QUALITY     NO         // select between YES , NO

// --------- 20 - Sequencer ---------                                               see oXs_config_advanced.h (only when oXs has to generate signals in sequence)

#endif// End define OXS_CONFIG_BASIC_h

ucakar, 2022-01-28:
Hi,

I measured VCC and was 3,3V so I assume I have 8Mhz and 3,3 V board.

But I selected and flashed as 8Mhz and 3,3 Volt just VFAS selected but still no sensor found :(

ucakar, 2022-01-28:
OK, I have 1 piece of Nano 33 BLE on stock.

Can i use it? Are pins the same?

https://docs.arduino.cc/hardware/nano-33-ble

mstrens, 2022-01-28:
It uses a different microprocessor so it will for sure not work.

ucakar, 2022-01-28:
I allready tested :)

So - my measurement 3,3V on VCC defines board as 8MHZ and 3,3Volts, correct?

mstrens, 2022-01-28:
Probably.
When the regulated voltage is 3.3V, there is 99% "chance" that it is a 8 mHz quartz on it.
An AVR 328P microprocessor is not guaranteed to work at 16Mhz with such a low voltage.

You can test it using a blink sketch (see example inarduino IDE).
If you flash the sketch using a 16Mhz board in arduino ide, you will get the led blinking twice slower (compare to the expected frequency based on the delay you put  in the sketch) if it a 8 mHz quartz.

ucakar, 2022-01-28:
So I have to find Arduino Pro mini or Nano with 16Mhz 5V and test it, I hope It will finally work :)

Regards from Slovenia

ucakar, 2022-01-28:
Should be that kind of alternaative with integrated micro USB OK?:

AArduino Pro Micro 5V/16MHz ATmega32U4 Replace Pro Mini ATmega328 For Arduino

https://www.ebay.com/itm/403310372269?hash=item5de72be5ad:g:W08AAOSw4mRebyT1

mstrens, 2022-01-29:
You link shows an arduino pro mini that uses a ATmega32U4.
I expect it should not work.

You need a Pro mini or (best because smaller than nano) or a nano. Both must have a AVR328p 5V 16Mhz.
Avoid the boards with a AVR168 (they have less memory and I never uses them so I am not 100% sure it should be OK)

You can find them on aliexpress searching with "arduino pro mini 5V 328".
Here the first link I get (but there a many others).
https://fr.aliexpress.com/item/1005003016788891.html?spm=a2g0o.productlist.0.0.680f3d41BopSTE&algo_pvid=f84f1a77-e7d2-44e9-93f8-68250ff99bc6&algo_exp_id=f84f1a77-e7d2-44e9-93f8-68250ff99bc6-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000023246916993%22%7D&pdp_pi=-1%3B4.02%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch.

or here
https://fr.aliexpress.com/item/32831029894.html?spm=a2g0o.productlist.0.0.680f3d41BopSTE&algo_pvid=f84f1a77-e7d2-44e9-93f8-68250ff99bc6&algo_exp_id=f84f1a77-e7d2-44e9-93f8-68250ff99bc6-7&pdp_ext_f=%7B%22sku_id%22%3A%2267224393175%22%7D&pdp_pi=-1%3B4.39%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch

Take care that some vendors sells severals model (3V/5V, AVR168/AVR328). Select 5V 16Mhz 328 to be safe.

ucakar, 2022-01-29:
Great, can you send me some link with same board, but micro SD port integrated?

ucakar, 2022-01-29:
Hi,

could be that board Arduino Nano V3.0 also compatible:

https://trgovina.svet-el.si/product/1elu0203/

It has already USB port soldered and is available here in Slovenia :)

mstrens, 2022-01-29:
The one you show is OK.

Otherwise you have plenty on aliexpress too
https://fr.aliexpress.com/wholesale?catId=0&initiative_id=SB_20220129080606&SearchText=arduino+nano

ucakar, 2022-01-29:
I agree, but i have to wait for it a few weeks :(

ucakar, 2022-01-31:
Hi,

It this board also compatible?

https://www.ebay.com/itm/264264910446?hash=item3d876a866e:g:zcoAAOSwvmxcokmJ

Thx,

Andrej

mstrens, 2022-02-01:
Yes, it should be compatible

ucakar, 2022-02-01:
OK, I like it because it has USB C connector :)

ucakar, 2022-02-02:
Hello,

I have to announce that I finally got working GPS on my old Arduino Pro Nano 328p 8Mhz 5V and that it is fully compatible with FrSky ACCESS recievers :)

Here is video of working GPS:
http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/openX_GPS.mp4

The problem was that I was using wrong S-Port on my FrSky R9SX ACCESS receiver.

Here is picture of problem:

http://www2.arnes.si/~aucaka1/modelarstvo/arduinopromini_GPS/r9sx.jpg

@mstrens Thank you again for all time and effort, I am really grateful!!! If I can anyway anytime can help you, I will be pleased.

Thanks again, Andrej, modeller from Slovenia

mstrens, 2022-02-02:
Fine that you solved your issue. Have fun with oXs.