openXsensor/openXsensor #32: Current sensor with ADS1115 and

MarcoRadloff, 2017-08-18:


mstrens, 2017-08-18:
Indeed, currently, I did not yet added those fields for Jeti protocol.
Current and consumption using arduino ADC should be OK.

If you need it, I can add it but I am not sure it is really requested by many users

F10F4F9, 2017-08-19:
would be very nice. 

By the way... 
Regarding the GPS module 
Is it possible to transfer the Travel distance and the coordinates?

mstrens, 2017-08-19:
I will look to add coordinates.

What do you mean by travel distance? Is it not possible to let this distance being calculated on Tx side if coordinates are transmitted?
Is it the cummulative distance covered since the startup?

If yes do you know how it is usually calculated. I expect that a new travel distance should be calculated only when the distance between last position and the "previously used " position exceed some value. Otherwise, due to gps unprecision, there is a risk that the travel distance increases even when the sensor does not move. 
I expect that there is also a need to reset the value without having to shutdown oXs. This would require to send to menu to the jetibox and to handle some keys press.

mstrens, 2017-08-19:
About gps coordinates, it seems that most of the code is already in oXs.
Probably that uncommenting some lines would let oXs send them.

I have no jeti and no gps to test it.
If you can test it, you just would have to uncomment following lines in oXs_out_jeti.cpp (start at line 728
      case GPS_LONG :                                          // Still to be added 
        mergeLabelUnit( textIdx, "Gps Long", "°"  ) ;
        break ;
      case GPS_LAT :                                           // Still to be added
        mergeLabelUnit( textIdx, "Gps Lat", "°"  ) ;
        break ;

mstrens, 2017-08-19:
I put a new version on github that should send V (instead of Volt), that should send current from ads1115 and gps coordinates

ZINKTiti, 2017-08-19:
As I m back home from my holidays I can test I have a DC16 and a GPS to test it. I can do it tomorow

F10F4F9, 2017-08-19:
Yes I have a Jeti TX and GPS sensor to test!
My be it is possible to calculate the distance/odometer with a Lua script on the TX side but I am not Sure. 

oXs can start calculating the Distance/Odometer if speed > say 10kph ... and reset on every powercycle

mstrens, 2017-08-20:
I still think that there was some missing code for GPS coordinates.
I put an update on git hub (version 8.2.2 now) but I can't test it.

F10F4F9, 2017-08-20:
Thank you I will test it in the next days.

F10F4F9, 2017-08-23:
Yesterday I had some time to check the current measurement over the ads1115.
The Current Telemetry Field is transmitted to the TX but there is no value inside.
On Debug I see the corresponding Voltage from the ads1115 but the TX says nothing.

this evening I will test the GPS coordinates.
I have start to write a small LUA script to calculate the Distance on TX side and it seems to be working.
If the openXsensor is sending the coordinates I can test it in the field not only on the desk.

mstrens, 2017-08-23:
I checked the code for current measurement with ads1115 and I did not see the bug (but I did not test)
Could you send me your config files in order to look if they are OK.
Can you also say me the type of current sensor you are using and all the resisors being used (e.g. do you use 2 resistors to make a voltage = Vcc/2 that you connect to one of the pin of ads1115?).
Can you give me the voltage that you get from ads1115 when you run in debug mode
If needed, I could add some debug options to the code in order to help to find the bug.

F10F4F9, 2017-08-24:
I use a Arduino Mini Pro 3,3V 8MHz with a ACS758 –U150A without any resistors direct connected to 3.3V and to the ads1115.
On the Debug I read from the ads 407mV for 0A and 442mV for about 2A.

[openXsensor_configs.zip](https://github.com/openXsensor/openXsensor/files/1247951/openXsensor_configs.zip)

mstrens, 2017-08-24:
In your config, you should change 
#define MVOLT_AT_ZERO_AMP              0.407   // in millivolt
to 
#define MVOLT_AT_ZERO_AMP              407   // in millivolt

Still this does not explain why you do not get data.
I made some changes in order to send some more data to the pc terminal when DEBUG is activated (in oXs_config_advanced.h)
In file oXs_ads1115.cpp, I added a parameter DEBUGCURRENT in order to let print the current and consumed milliAmp each time those values are calculated.
In file oXs_out_jeti.cpp, I added a parameter DEBUGADSCURRENT in order to let print the current and consumed milliAmp each time those values are put in some fields that should be send to the Tx.

Could you test this.

[openXsensor.zip](https://github.com/openXsensor/openXsensor/files/1250030/openXsensor.zip)

F10F4F9, 2017-08-25:
I test it shortly this morning and there are no current output from the ADS1115 “DEBUGCURRENT”.

mstrens, 2017-09-02:
I will check it.

onki69, 2017-09-05:
The track lenght calculation is described here:
https://en.wikipedia.org/wiki/Great-circle_distance

I also want to use the ADS1115 for current measurements since it is the most reliable method when using shunt based sensors (like the current sensors from SM-modellbau etc.).
I will setup such a sensor shortly an can test it with my Jeti equipment.

mstrens, 2017-09-05:
I found a bug when oXs calculates the current using an ads1115.
It could explain why there was no output from DEBUGCURRENT.
I put a new version on github to fix it but I did not tested it.
Please test it.

F10F4F9, 2017-09-05:
I will test it this evening. 
Thanks for you great support!

F10F4F9, 2017-09-12:
Sorry for the delay.
I have some output on DEBUGCURRENT & DEBUGADSCURRENT. 
Next step is to check it on the TX side and see if i get the values that I expect.

F10F4F9, 2017-09-13:
I get the values for Current & Consumption but i have to change the unit in the oXs_out_jeti.cpp from “Amp” to “A” & "AmpH" to "mAh" otherwise there is no voice for the unit from the Transmitter only the value.
The values that i get for Current & Consumption seem to be correct i have a error on my side.
I power the ACS758 with 3,3V and i have to check the sensitivity mv/A

mstrens, 2017-09-14:
Ok thanks. I will change the units in next version on github.

closed 2017-09-19 by F10F4F9