openXsensor/openXsensor #38: RPM not working with Jeti protocol

onki69, 2017-09-11:
I have tested it in several configurations but I never got any output neither a sensor parameter when I set
`#define CALCULATE_RPM     YES` when using Jeti protocol.
Since Jeti does not offer any calculation possibilities at the TX it would also be nice to get an additioal parameter for the correct calculation of the value (props, gear, motor poles).

Would be great if someone could check the code because I am not good enough in programming to understand the code :-).
Thanks and best regards
Onki

mstrens, 2017-09-14:
Indeed, the current version of oXs does not contains code to transmit RPM.
Jeti protocol has been added lately and  I mainly coded the data asked by a friend.
I can add RPM in the next release (with a parameter for the correct calculation).
Do you know the code to be as unit in order to get also vocal announcement.

mstrens, 2017-09-14:
New version is uploaded on github to fix this.
Still I did not test it.

onki69, 2017-09-18:
RPM works on a little test setup using a YEP ESC rpm output pin.
Thanks for implementing this.
Do you think it makes sens to tag items who are not yet implemented for Jeti (or other protocols with a letter (or a combination to seen which protocols are supported)? Something like (FH,FS,M,H,J)?
It seems to me that pushbutton and/or PPM reset (flow sensor) is not working with Jeti either.

closed 2017-09-18 by onki69

mstrens, 2017-09-19:
I do not really remember which items are not implemented for some protocols.
Looking at it will nearly take the same time as implementing the item.

For flow sensor in Jeti protocol, I implemented extra features that allow from the Tx side ( using the button from the jetibox ) to
- force a reset
- define the tank capacity
- to fill 4 sets of calibration values
This works for sure. I saw it on the Tx from a friend.

About pushbutton and/or PPM reset, I do not think that there is specific code in oXs for Jeti protocol. If it does not work for jeti, it should probably also not work for other protocols.
Still, so far, I did not yet get an issue about it. Probably it is not often use because on Tx using a firmware like openTx:
- battery consumption is calculated on Tx side ( and so reset is done on Tx)
- fuel consumption (from flow sensor) can be done with a lua script without using PPM