openXsensor/openXsensor #41: update for NMEA GPS?

guerradr, 2017-12-01:
It would be great to implement NMEA GPS support, since these are readly available and super cheap. The sizes of the non u-blox typ gps are most of the times smaller.

F10F4F9, 2017-12-04:
This one is mini and works perfect and is cheep.  

Mini GPS For NAZE32 https://banggood.app.link/j2WiDEm4AI

mstrens, 2017-12-30:
Decoding the NMEA messages would require more process and more memory on oXs side.
Decoding UBLOX messages is faster and requires less memory.
In order to support more concurrent functionalities in oXs, it is better to use UBLOX messages.
It seems that the gps from the link above use a ublox chip. So I expect it should work with oXs.
Best is to check with the program U-center provided by ublox if it can really be reconfigured to generate ublox message.
To work with oXs, the gps must be configured (e.g. using U-center)  in order that it accepts NEMA commands at 9600 bauds. This is the default config for a ublox gps but it could be that the module you linked has been configured in a different way. 
If the module accept commands at 9600 bauds, oXs will (at each power on) send some commands to ask for Ublox messages at 38400 bauds.

closed 2017-12-30 by mstrens