openXsensor/openXsensor #63: S.Port: Delay before send is missing

RetoD7, 2020-10-05:
oXs is not working with new FrSky Archer (2.1.2). But it makes also troubles with other FrSky RX. The delay (400us) is missing befor data sending.
OCR1A should not be set at beginn of RECEIVE state. This should be done only for the next bit.

its working (on Arduino Pro Mini 3.3V 8MHz):
```
    case RECEIVE :  // SPORT ****  Start bit has been received and we will read bits of data receiving, LSB first.    
           {
                    uint8_t data ;        // Use a temporary local storage
                    data = SwUartRXBitCount ;
                    if( data < 8 ) {                         // If 8 bits are not yet read
                        OCR1A += TICKS2WAITONESPORT ;                    // Count one period after the falling edge is trigged.
                        SwUartRXBitCount = data + 1 ;

```

RetoD7, 2020-10-06:
With the above change, it's working okay on Archer and other RX, when there is no gps connected.

With a gps connected it's working some time (1-10 minutes?) and then all oXs sensors are lost. Sometime it's recovering itself, but this takes time (5-20 minutes?). It's some kind of deadlock.

Without my change it's working more or less also with gps but not on Archer RX. But on the oscillograph there are collisions (undefined levels) visible, because of the missing delay. (on Archer and other RX)

Do I overlook some dependency?
Is it a problem of IRQ timing and priority? (I have no experience about this on arduino yet.)
Or could it be a performance problem of the 8MHz adruino?

RetoD7, 2020-11-01:
It's a performance problem with the 8MHz Arduino. At least in combination with a GPS sensor it's not working correctly.

The 16MHz Arduino is working, as far I can test with the small 168-variant.

Hobby4life, 2020-11-10:
I having same experience/behaviour with G-RX6 and RX6R with ACCESS fw and latest Archer receivers and firmware up to v2.1.5

A lot of my sensors don't work anymore.

RetoD7, 2020-12-08:
**Everything is working perfectly with an 16MHz Arduino (Pro Mini 328).** 

Beside the initial problem that the "8MHz"-openXsensor is not working with an archer receiver, there are also many telemetry errors caused with other FrSky receivers. (RX8 pro, etc.) So it's a general performance problem of the 8MHz version. I don't know if the problem also depends on compiler or compiler options.

But for me the problem is solved with the 16MHz Arduino.

ucakar, 2022-01-27:
Hi I am using Archer GR8 and I can not find any sensor?? I am using ACCESS, and Arduino pro mini, 16 mhz 328, 5V board.

gbe1200, 2022-12-25:
I am busy already to get it work for 4 days but it is impossible getting it to work, transmitter is not finding the sensors, can saomeone help me please, i don´t know anymore what to do. Or someone has the complete files for me that work on there x8r, thanks in advance