openXsensor/openXsensor #64: GPS is not working

labeh, 2021-02-25:
I try to built an openXsensor for my Frsky Horus X12s with X6R receiver an ublox neo-m8n GPS. For testing purpose I use an Arduino nano. Later on I would like to transfer the program to an Arduino mini.

GPS RX + 10k Res. > Arduino TX
GPS TX > Arduino RX
Frsky Sport > Arduino D4

To be sure that the GPS is working properly I tried it with the ublox u-center without any issues. The RX LED on Arduino is constantly blinking with ca. 1HZ frequency. I tried to connect to the Arduino console and to enable debugging, but I haven't seen any output yet. I copied the configuration files from the example with the GPS, current sensor, etc. but this does not work either for me. It looks like the initialization of GPS does not work as expected.

labeh, 2021-02-25:
I now managed to connect to the debugging and get the following output:

`02:25:52.875 -> openXsensor starting..
02:25:52.875 ->  milli=0
02:25:52.875 -> freeRam=1115
02:25:52.875 -> Current sensor on pin:6
02:25:52.875 -> Reference voltage:5000
02:25:52.908 -> Offset for current:511.50
02:25:52.908 -> mAmp per step:81.46
02:25:52.908 ->  milli=2
02:25:58.834 -> End of GPS setup
02:25:58.834 -> FRSky Output Module: TX Pin=4
02:25:58.834 -> Sport protocol= 1
02:25:58.867 -> End of general set up
02:25:58.867 -> in loop=5944
02:25:58.867 -> read gps at 5949
02:25:58.867 -> call calculateAllFields at5949
02:25:58.867 -> end of call calculateAllFields at5951
02:25:58.945 -> in loop=5955
02:25:58.945 -> read gps at 5959
02:25:58.945 -> call calculateAllFields at5960
02:25:58.945 -> end of call calculateAllFields at5961
02:25:58.945 -> in loop=5965
02:25:58.945 -> read gps at 5970
02:25:58.945 -> call calculateAllFields at5970
02:25:58.945 -> end of call calculateAllFields at5972
02:25:58.945 -> in loop=5975
02:25:58.945 -> read gps at 5980
02:25:58.945 -> call calculateAllFields at5980
02:25:58.945 -> end of call calculateAllFields at5982
02:25:59.025 -> in loop=5986
02:25:59.025 -> read gps at 5991
02:25:59.025 -> call calculateAllFields at5991
02:25:59.025 -> end of call calculateAllFields at5993
02:25:59.025 -> in loop=5996
02:25:59.025 -> read gps at 6001
02:25:59.025 -> call calculateAllFields at6001
02:25:59.025 -> end of call calculateAllFields at6003
`

the last bit is repeated all the time again. But still no GPS sensor available during Sensor discovery.

onki69, 2021-02-26:
Did you test this with a GPS fix at the Sensor?
I am not sure but if I remember correctly the GPS is only recognized when the GPS has a valid fix.
It might be difficult to get this indoor but just give it a try.