openXsensor/openXsensor #67: Incorrectly sending negative current to FrSky

MikeBland, 2021-07-12:
The value of current sent using FrSky protocol may be negative. This is not valid for the FrSky telemetry protocols that only support positive values for current.
Recent radio OSs (e.g. Ethos) reject negative values and this may result in sensor loss.

Hobby4life, 2021-07-13:
Correct, this is the only sensor I have seen so far that have "odd" values sent over smartport.

I opened this case on the Ethos bug tracker:
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/585

Hobby4life, 2021-07-13:
I have added a quick fix in:

https://github.com/openXsensor/openXsensor/blob/master/openXsensor/oXs_out_frsky.cpp

`#if defined(ARDUINO_MEASURES_A_CURRENT) && (ARDUINO_MEASURES_A_CURRENT == YES)  
    if ( oXs_Current.currentData.milliAmps.available) {
      oXs_Current.currentData.milliAmps.available = false ; 
      sport_currentData.value = oXs_Current.currentData.milliAmps.value  / 100 ;
      sport_currentData.available = true ;
    }  
#elif defined(AN_ADS1115_IS_CONNECTED) && (AN_ADS1115_IS_CONNECTED == YES ) && defined(ADS_MEASURE) && defined(ADS_CURRENT_BASED_ON)
    if ( oXs_ads1115.adsCurrentData.milliAmps.available ) {
      oXs_ads1115.adsCurrentData.milliAmps.available = false ;
      sport_currentData.value = oXs_ads1115.adsCurrentData.milliAmps.value  / 100 ;
      sport_currentData.available = true ;
    }  
#endif`

to

`#if defined(ARDUINO_MEASURES_A_CURRENT) && (ARDUINO_MEASURES_A_CURRENT == YES)  
    if ( oXs_Current.currentData.milliAmps.available) {
      oXs_Current.currentData.milliAmps.available = false ; 
      sport_currentData.value = oXs_Current.currentData.milliAmps.value  / 100 ;
      if (sport_currentData.value  < 0){
        sport_currentData.value = 0;
      }
      sport_currentData.available = true ;
    }  
#elif defined(AN_ADS1115_IS_CONNECTED) && (AN_ADS1115_IS_CONNECTED == YES ) && defined(ADS_MEASURE) && defined(ADS_CURRENT_BASED_ON)
    if ( oXs_ads1115.adsCurrentData.milliAmps.available ) {
      oXs_ads1115.adsCurrentData.milliAmps.available = false ;
      sport_currentData.value = oXs_ads1115.adsCurrentData.milliAmps.value  / 100 ;
      sport_currentData.available = true ;
    }  
#endif`

romoloman, 2022-02-26:
I have added your fix to my own fork.
I have done a pull request for openxsensor that includes this and other fix/improvements.

PR #71 (2022-03-07): Various improvements to openXsensor and it's configurator.
I have completed oxsconfigurator,
In openxsensor i have:
1) Added the possibility of trasmitting the temperature of vario sensor
2) Transmit date and time from GPS (sport only)
3) Configure BMP280 Address
4) Removed negative values from current
fixes #67  fixes #68 fixes #65