openXsensor/openXsensor #68: Request: Add GPS Date & Time (smartport)

Hobby4life, 2021-07-19:
I like to request date and time from GPS

romoloman, 2022-02-25:
Hi I have implemented the request in my own fork.
Have a look and let me know if it works for you.

Hobby4life, 2022-02-25:
Will let you know.. will test tonight :-)

romoloman, 2022-02-25:
To enable the feature you can generate the config using the configurator or define GPS_TRANSMIT_TIME in osx_config_advanced.h

ParkerEde, 2022-02-25:
Hi romoloman,
sounds great. Where can I get your fork to test it?
Will you also implement it in the regular repo?

romoloman, 2022-02-25:
click on my icon and have a look to my repositories. You will find openXsensor.
I have done several modifications to openXsensor expecially in the oxsconfigurator and also to the vario (to be able to get air temperature) I don't know if it's a good idea to push commit in regular repository.

romoloman, 2022-02-25:
BTW I plan to add date/time also to sensor hub telemetry by implementing frame3 packet.
(i still have some D8 receiver around)

ParkerEde, 2022-02-25:
it works perfect. BIG Thank you. 
One additional question. Does it make sense to reduce the transmission interval for DateTime on the sport to once per second?

Hobby4life, 2022-02-25:
That would indeed be better. Dont know the timeout limit for date and time. But as long as possible intervals.

romoloman, 2022-02-25:
well i have vario, voltages, gps, in my case It's already less than 1 second but we can reduce the 1 per second by keeping previous trasmitted value and transmit it only when value change (so 1 per second for time and 1 per day for date).

Hobby4life, 2022-02-25:
Interval of 1sec for now is fine for both Date and Time.

romoloman, 2022-02-25:
Committed... please check.

ParkerEde, 2022-02-25:
with the new version, I got no DateTime anymore.

#define GPS_TRANSMIT_TIME is active in oXs_config_advanced.h

romoloman, 2022-02-25:
ok i will check this night with a sensor available...

Hobby4life, 2022-02-25:
I have tested your first firmware, and the interval there is very acceptable for me :-) 500mS?
And works with ETHOS ;-)

romoloman, 2022-02-25:
within one hour or two I will be able to make some test with the sensor,
Probably my last commit should work, but i'm not sure.
Previous one was broken as I did it while doing something else :(.....

Hobby4life, 2022-02-25:
> within one hour or two I will be able to make some test with the sensor, Probably my last commit should work, but i'm not sure. Previous one was broken as I did it while doing something else :(..... BTW did you testes also oxsconfigurator ? I have implemented almost all the tabs including sequencer.

I have seen it added the GPS TIME definition, so thats works.. but did not compiled with the generated files.
I have manually added the definition to my configs.

romoloman, 2022-02-25:
Ok latest commit slow down TIME and date trasmission, and it works.

ParkerEde, 2022-02-25:
Fantasitic. It works great. I added manually the line
#define GPS_TRANSMIT_TIME 
to my oXs_config_advanced.h.
I've done one test with the Windows Configurator, this works well.

All my GPS Sensors are already flashed now with your fork. Very very thank you.

Kind regards
ParkerEde

romoloman, 2022-02-25:
Hi ParkerEde, have a look to the latest commit, it avoid a stupid call to millis() so maybe your GPS_TRANSMIT_TIME becomes useless.
Calling millis() uses a lot of bytes, the last commit instead set data available only when seconds changes so once per second, and that's enough.

ParkerEde, 2022-02-25:
okay, thank you. I applied the last commit. Looks good. My GPS Sensors are now updated.

romoloman, 2022-02-26:
"I also had a fix regarding negative current over smartport. ETHOS does not like this."
It's already in my fork, as well a fix regarding BMP280

PR #71 (2022-03-07): Various improvements to openXsensor and it's configurator.
I have completed oxsconfigurator,
In openxsensor i have:
1) Added the possibility of trasmitting the temperature of vario sensor
2) Transmit date and time from GPS (sport only)
3) Configure BMP280 Address
4) Removed negative values from current
fixes #67  fixes #68 fixes #65
closed 2022-03-07 by mstrens