openXsensor/openXsensor #72: FBUS telemetry support

Hobby4life, 2022-02-27:
Now Frsky has added telemetry to their FBUS system it would be nice to have FBUS telemetry implemented to OpenXsensor :-)

romoloman, 2022-02-27:
I suppose you mean FPORT(2), 
Maybe it can be done but I suppose that ATMEGA 328 even at 16 MHz is too limited for handling timing.
Personally I lack fport hardware to make tests as I still rely on ACCST for all my models.

Hobby4life, 2022-02-27:
> I suppose you mean FPORT(2), Maybe it can be done but I suppose that ATMEGA 328 even at 16 MHz is too limited for handling timing. Personally I lack fport hardware to make tests as I still rely on ACCST for all my models.

FPORT 2.0 has been renamed to FBUS some while ago :-)

Maybe a WEMOS D1 + OTA firmware loader, would be a nice platform to start with :-)

romoloman, 2022-02-27:
WEMOS D1(8266) has few I/O ports, maybe a Teensy LC can be the way to go, three H/W UART, port inversion.
I suppose that's enough, maybe too much expensive.

StefanKellerAC, 2022-02-28:
RP2040 would be fine, I managed to invert a Port simply from C++, no PIOASM is needed
I just needed a hardware serial, both Pins connected together, and this code in Pawelskys S-Port Library:

    PIN_TX = 4;
    PIN_RX = 5;
    Serial2.setTX(PIN_TX);
    Serial2.setRX(PIN_RX);

```
  void FrSkySportSingleWireSerial::setMode(SerialMode mode)
  {
    if(port != NULL) 
    {
      if(mode == TX) 
        {
        //Serial.println("tx");
        gpio_set_inover ( PIN_RX , GPIO_OVERRIDE_HIGH );  // disable Input to idle  
        gpio_set_oeover(PIN_TX, 0 );  // re enable output

        gpio_set_outover(PIN_TX, GPIO_OVERRIDE_INVERT);  // enable invertet output
        }
      else if(mode == RX)
        {
        //Serial.println("rx");
        gpio_set_oeover(PIN_TX, GPIO_OVERRIDE_LOW );  // GPIO_OVERRIDE_LOW is disable output
        gpio_set_inover ( PIN_RX , GPIO_OVERRIDE_INVERT );  // enable Input
        };
    }
  }

```

a RP2040 Zero is really small, for just  4.99$
[image]

https://www.waveshare.com/rp2040-zero.htm

romoloman, 2022-02-28:
I didn't noticed RP2040 Zero, but indeed is small and cheap, the price is comparable with latest arduino promini
Is it compatible with pico ? (i suppose yes as they share the same chip)

StefanKellerAC, 2022-02-28:
yes, only the pins differ a bit. It has a WS2812 onboard, all 4 ADCs are available, ...

romoloman, 2022-02-28:
And for the locator is also interesting the version with integrated TFT display (also available on waveshare website)
Tomorrow I will receive the pico version, I will have something to play with, 
Furthermore considering the resources available it will be possible also to integrate sbus2 and other protocols.

romoloman, 2022-03-02:
I'm start playing with raspberry pico.
The nice thing is that is very easy (even in the arduino port) to have two task running on the two cores, one for handling communication with the RX and the other to read sensors and perform calculations.
We can have also sensors using single wire that depend on microdelays for handling communication (ie DS1820)
Maybe a new branch ? Or a new project ? (I cannot understand how much alive is openxsensor)

StefanKellerAC, 2022-03-03:
OXS depends heavily on specific processor stuff, just try to compile it for RP2040 without compile errors! you have to comment out a lot of stuff.
I will do it with Pawelskys lib :-)