openXsensor/openXsensor #78: Support the GPS of UBLOX M10

ahfynjj, 2023-04-02:
After a simple test, the GPS performance of the M10 chip has been greatly improved compared with the M8, but the oXs cannot be recognized. I have failed to configure this GPS many times. Can you add support for the new M10, thank you .

romoloman, 2023-07-23:
Added in https://github.com/romoloman/openXsensor
See also pull request from me.

romoloman, 2023-07-27:
@mstrens Thanks. Issue should be solved.

ahfynjj, 2023-11-02:
thank you so much

ShawnLKY, 2024-03-09:
Looking to connect the UBLOX ZED-F9P GPS, will setting the M10 option allow for data logging?

mstrens, 2024-03-09:
What do you mean by M10 option?

About data logging, do you think about logging on the handset or using the oXs logger?

ShawnLKY, 2024-03-09:
I see that the M10 and M8 uses different baud rate. 

The F9P uses 38400 which is the same as the M10.
Can i set it as a M10 to acquire the GPS data. 

I plan to log on both the transmitter and an onboard SD.

mstrens, 2024-03-09:
In principe, oXs try to configure the baudrate automatically for M10.
If your GPS can keep the configuration after a shutdown, then you can also configure it manually and say in the parameters of oXs that gps config is external.

ShawnLKY, 2024-03-09:
Excellent. i will test it out. 

Thank you!

mars-wolf, 2024-06-13:
Can please someone help me and send my an *.ucf file of the M10 for the u-center2 Software? My M10 is working on u-center2, gets GNSS fix but unfortunalety I have no clue about the settings in u-center2.
If someone could read the settings that I can upload them on my GPS-Hardware I would be very, very happy! Thank you!