openXsensor/openXsensor #80: Lot of sensor lost alarm

Minos74, 2023-08-15:
Hi,
I have a X20s on FW1.4.9 with Archer R4 and openXsensor.

With my fisrt model it works perfectly fine, and same configuration on another model and I have plenty sensor lost alarms on diverse sensors

Attached the log of the last flight

[KRAKEN-700-2023-08-13-09-39-01.csv](https://github.com/openXsensor/openXsensor/files/12352672/KRAKEN-700-2023-08-13-09-39-01.csv)

Any idea where to investigate ?
Thank you for your help
Minos

romoloman, 2023-08-17:
How is openxsensor configured ?
Are you using it with other sensors connected ?

Minos74, 2023-08-17:
Hi,
No other sensors exept the Frsky internal ones.
OXS is the lastest version 8.0.1

Here is my configuration

// OpenXsensor https://github.com/openXsensor/
// started by Rainer Schlosshan and maintained by Michel Strens

// This is version : 8.02 (19 february 2017)

//******************************************************************************************************************************************************* //
//                                                                                                                                                        //
// ! IMPORTANT NOTE !!!!!!!!!!   Another file in this project provides detailed explanations on how to set up this file (see oXs_config_description.h).   //
//                                                                                                                                                        //
//******************************************************************************************************************************************************* //
//  Note: the oXs_config.h file present on the GitHub site is not always meaningful.
//       It could be that the combination of active/non active parameters is not consistent.
//       This is just the result of many updates and tests in this document.
//       So take always care to set up the oXs_config.h file according to your needs and check carefully all options.
//  For some previous version of oXs, there was a program named "configurator" running on PC that allowed easier editing of this oXs_config.h file
//       Still this configurator is not developped anymore. So currently, this oXs_config.h file has to be edited manually. 

#ifndef OXS_CONFIG_h
#define OXS_CONFIG_h

// --------- 1 - Telemetry protocol ---------
#define PROTOCOL  FRSKY_SPORT  // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

// --------- 2 - Serial data pin choice ---------
#define PIN_SERIALTX      4    // The pin which transmits the serial data to the telemetry receiver, Usually pin 4 (otherwise pin 2)

// --------- 3 - PPM settings ---------
//#define PIN_PPM           3       // default is 2 but my own device use 3
#define PPM_MIN_100       980     // default 1500 - 512 ; // pulse width (usec) when TX sends a channel = -100
#define PPM_PLUS_100      1990    // default 1500 + 512 ; // pulse width (usec) when TX sends a channel = +100

// --------- 4 - Vario settings ---------

// ***** 4.1 - Connecting 1 or 2 MS5611 barometric sensor *****
//#define VARIO // set as comment if there is no vario
//#define SENSOR_IS_BMP180 // set as comment if baro sensor is MS5611 and BMP180 or BMP085 is used
//#define SENSOR_IS_BMP280 // set as comment if baro sensor is MS5611 and BMP280 is used
//#define VARIO2 // set as comment if there is no second vario

// ***** 4.2 - Sensitivity predefined by program *****
#define SENSITIVITY_MIN 50
#define SENSITIVITY_MAX 300
#define SENSITIVITY_MIN_AT 100
#define SENSITIVITY_MAX_AT 1000

// ***** 4.3 - Sensitivity adjusted from the TX *****
#define SENSITIVITY_MIN_AT_PPM 10    // sensitivity will be changed by OXS only when PPM signal is between the specified range enlarged by -5 / +5
#define SENSITIVITY_MAX_AT_PPM 40
#define SENSITIVITY_PPM_MIN  20      // common value for vario is 20
#define SENSITIVITY_PPM_MAX 100      // common value for vario is 100

// ***** 4.4 - Hysteresis parameter & Alt compensation based on sensor temp *****
#define VARIOHYSTERESIS 5
//#define ALT_TEMP_COMPENSATION 800

// ***** 4.5 - Vertical speeds calculations *****
#define VARIO_PRIMARY       FIRST_BARO          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define VARIO_SECONDARY     FIRST_BARO          // select between FIRST_BARO, SECOND_BARO , AVERAGE_FIRST_SECOND, AIRSPEED_COMPENSATED , BARO_AND_IMU
#define SWITCH_VARIO_MIN_AT_PPM 10
#define SWITCH_VARIO_MAX_AT_PPM 90

// ***** 4.6 - Analog vertical speed *****
//#define PIN_ANALOG_VSPEED 3
#define ANALOG_VSPEED_MIN -3
[... 148 lines cut ...]
  int32_t value ;
} ;

#define FRSKY_SPORT      1
#define FRSKY_HUB        2
#define FRSKY_SPORT_HUB  3
#define MULTIPLEX        4
#define HOTT             5
#define JETI             6

#define SECONDS_SINCE_T0        32
#define AVERAGE_VSPEED_SINCE_TO 33 

#ifdef DEBUG
//#include "HardwareSerial.h"
#endif

#ifdef GPS_INSTALLED
//#include "HardwareSerial.h"
#endif

#endif// End define OXS_CONFIG_h

Regards
Minos

romoloman, 2023-08-17:
Try to download source end not the release and generate the config file using oXs_Configurator.

[image]

Minos74, 2023-08-18:
Thank you I will try.
You are right I have current, voltage and RPM also
Minos